ARMmbed/mbed-os-example-ble #347: missing file: pretty_printer.h

factoryal, 2020-12-03:
### Description of defect
`pretty_printer.h` file missed on this 6.5.0 release. `02f839a`
cannot run example code since that file not found on compiler.
```
[Fatal Error] main.cpp@20,10: 'pretty_printer.h' file not found
[ERROR] .\source\main.cpp:20:10: fatal error: 'pretty_printer.h' file not found
#include "pretty_printer.h"
^~~~~~~~~~~~~~~~~~
1 error generated.
```

#### Target(s) affected by this defect ?
Nucleo-WB55

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
example: `02f839a`
mbed-os: 6.5.0 `f227856`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.3.1

#### How is this defect reproduced ?
Always

Thanks.

factoryal, 2020-12-03:
problem you mentioned has been fixed

paul-szczepanek-arm, 2020-12-03:
Pretty printer is part of mbed-os-ble-utils. This should be checked out by running `mbed deploy` inside the chosen example.

factoryal, 2020-12-05:
oh, so it was changed since this version?
I found that there's `pretty_printer.h` right on the previous version but now it's not.
I'll try out example once more.

thanks!

paul-szczepanek-arm, 2020-12-08:
Yes, it's been refactored out into a common repo as it's used by all examples. Please reopen if there's still an issue.

closed 2020-12-08 by paul-szczepanek-arm

idea--list, 2021-02-14:
@paul-szczepanek-arm 

> Pretty printer is part of mbed-os-ble-utils. This should be checked out by running mbed deploy inside the chosen example.

This arises 2 questions:
1. How should people get informed about this?? Especially as there is not a word about that throughout the documentation neither [here](https://os.mbed.com/docs/mbed-os/v6.7/apis/ble.html) nor [here](https://os.mbed.com/docs/mbed-cordio/19.02/introduction/index.html). Not to mention [this page](https://os.mbed.com/docs/mbed-cordio/19.02/introduction/index.html#getting-started) is stated to be a getting started page, but the content is absolutely useless. I mean 100 millions use Bluetooth daily and everyone knows that it is a wireless technology to send information. There is no need telling that. Then at the very bottom of all the bull**** one can find a link to the [mbed-os-example-ble repo](https://github.com/ARMmbed/mbed-os-example-ble), which again does not tell a word about the need of mbed-os-ble-utils, nor does that repo do any onboarding despite it is the only useful content on the getting started page. Going one step further in that repo it turns out, the examples are OUTDATED (yet again) and it simply does not even compile. This is very bad user experience.
2. What if someone uses the online compiler or Mbed Studio?? How should they deploy mbed-os-ble-utils?
3. Without any real explanations/tutorials how do you think anyone could understand how to use BLE with Mbed? For sure BLE engineers already know the ins and outs but exactly that is why they do not even need a documentation.

I hope all these make sense and there will be a tutorial that really onboards people with current and working examples.

paul-szczepanek-arm, 2021-02-15:
Hi.

1. There is no need to mention libraries by name in the readme, all you have to do is type `mbed deploy` in the chosen example (which the readme mentions). All libraries will be automatically checked out.

2. The online compile automatically checks out the libraries. I can see that the online compiler versions of examples are old. Thank you for bringing this to my attention. Our friends at ST have updated one of the examples here https://os.mbed.com/teams/ST/code/mbed-os-example-ble-Advertising/ . It uses the new utils library which checks out fine.

3. The examples are meant to teach you how to use BLE. They are an explanation. They idea is you clone this repo, compile the examples, run them and read the heavily commented code to see how it's done.

idea--list, 2021-02-15:
Now at least i know what the problem is: the mindset.

1. There is really no need to mention important info anywhere as people who are intended to make use of BLE already know that. mbed deploy is not an option in Mbed Studio, but again people should not use that toy. Sorry for being sarcastic but i can not help after such an answer.

2. Checked out the link you mention. That is implemented in Mbed OS V5.15.4... which is kind of outdated these days and my target is supported only since V6.3. So after importing that to Mbed Studio i updated to Mbed V6.7, compiled, flashed but it still does not run on my target.

3. That idea works only for people who are already familiar with BLE but as i mentioned before those people do not need any documentation. Let us just remove the current documentation as even that has more sense than letting people read that in it's current state.

paul-szczepanek-arm, 2021-02-16:
The utils library is not mentioned because it's an implementation detail and has nothing to do with BLE.

Mbed studio will check out the missing libraries for you. Open a folder with an example and it will show you libraries that are part of the example. It shows mbed os and the utils libraries and the fact they are missing. There's a button the arrow is pointing to that lets you fix the problem by checking them out for you.

[image]
[image]

What is your board?

idea--list, 2021-02-16:
Thanks for the explanation, i did not know Studio would check out missing libraries automatically.

My primary board is a [Sparkfun Artemis Thing Plus](https://www.sparkfun.com/products/15574) (love it because the small form factor, the radio is also FCC and CE certified, consumes only ca 6-8 mA while transmitting, etc). Sparkfun made up the Mbed core based on Ambiq SDK (is a HAL). Interestingly there is a BLE example for this board in the SDK so i guess that works. There is also an Arduino core (actually a hybrid Mbed-Arduino core) and with Arduino BLE library even under Arduino does this feature work.

For some reason however with Mbed Cordio the BLE feature does not work and as i am new to BLE i can not even try to debug it.

My secondary board is a MAX32630FTHR that has a dual-mode BT module which was once functional under Mbed, but it got broken long time ago and later Mbed ceased BT support for this target. Since yesterday however i am playing around with [this](http://bluekitchen-gmbh.com/) on the MAX board and i achieve ~230 KB/s payload. It seems that this stack is compatible with a bunch of different chips, and for non-commercial purposes it is also free to use. With recent updates this board also seems to resurrect under Mbed.

factoryal, 2021-02-16:
@paul-szczepanek-arm Hi.
I followed [Building and flashing examples](https://github.com/ARMmbed/mbed-os-example-ble#building-and-flashing-examples) but didn't worked since `mbed deploy` does nothing.
So I Just did copy-pasting the sources of one of examples.
I think it caused the missing problem.

I found that from mbed-os-6.5, it needs mbed-cli [v2](https://os.mbed.com/docs/mbed-os/v6.7/build-tools/mbed-cli-2.html) instead of [v1](https://os.mbed.com/docs/mbed-os/v6.7/build-tools/mbed-cli-1.html).
the execution command changed to `mbed` to `mbed-tools`.
When I executing this command,
```sh
mbed-tools deploy
```
instead of 
```sh
mbed deploy
```
the deployment worked and successfully compiled.

Please update guide.
Thanks.

paul-szczepanek-arm, 2021-02-16:
Old tools still work fine. Did you call mbed deploy inside a single example directory (not the root of the repo)?

factoryal, 2021-02-16:
@paul-szczepanek-arm 
Yes. I called deploy inside a single example directory.
The error is that there's no 'mbed' as an execution unit.
It seems that the name changed `mbed` to `mbed-tools` on mbed-cli v2.

paul-szczepanek-arm, 2021-02-16:
The name did not change. mbed-tools is the new tool that uses cmake to build mbed. The old ones are very much still alive and supported. Can you please give me the version of your mbed?

paul-szczepanek-arm, 2021-02-16:
`mbed --version`

factoryal, 2021-02-16:
@paul-szczepanek-arm 
7.0.0
[image]

paul-szczepanek-arm, 2021-02-16:
OK, I don't speak Korean but I'm guessing this is a windows error and not a mbed error

paul-szczepanek-arm, 2021-02-16:
Are you sure you even have mbed installed?
`pip install -U mbed-cli`

factoryal, 2021-02-16:
@paul-szczepanek-arm 
yes
[image]

factoryal, 2021-02-16:
@paul-szczepanek-arm 
are mbed-tools and mbed-cli different?

paul-szczepanek-arm, 2021-02-16:
Well, there's something wrong with your installation or there's a problem with powershell - the error is not coming from mbed, it's from windows. You should open an issue with https://github.com/ARMmbed/mbed-cli guys - they will know why the windows installation doesn't pick up the executable.

The new tools are different, yes. They're still an early development, they allow you to build mbed with cmake.

factoryal, 2021-02-16:
@paul-szczepanek-arm
https://os.mbed.com/docs/mbed-os/v6.7/build-tools/install-or-upgrade.html
it says "Mbed CLI 2 is a Python package called mbed-tools, so you can install it with pip."
I'm beginner in mbed programming and there's lot of things to know... um... just lot of confusion.
right I'm here is very midnight. so I'll check it later

Thanks for your support.

paul-szczepanek-arm, 2021-02-16:
They're both python libraries and AFAIK they use some common tools so likely mbed-tools deploy and mbed deploy does the exact same thing. It just looks like there's a problem with your installation of mbed-cli and powershell just can't find the executable although honestly I don't know how it works on windows - I'm sure the tools team can give you a hand with that.