ARMmbed/mbed-os-example-ble #351: BLE_GattServer_AddService does not work on NRF52-DK

apcurtiss, 2021-01-04:
### Description of defect
This looks a lot like issue #339, which I'm not sure was resolved. I've updated here with any new details:

The BLE_GattServer_AddService example compiles ok. However when using a BLE scanner and then connecting to the "Heartrate" device it disconnects unexpectedly, immediately after establishing a connection.

I added a disconnect reason printout to the disconnect callback and the error is as was reported in the initial bug report (0x08, timeout).

The nrf connect log says:

`Connected.`  
`Discovering Services...`  
`peripheral.discoverServices(nil)`  
`[Callback] centralManager(central, didDisconnectPeripheral: peripheral, error: The connection has timed out unexpectedly.)`

#### Target(s) affected by this defect ?
NRF52_DK

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.4 (macintosh)
nrf connect 2.4.6 (iOS)

#### How is this defect reproduced ?
compile BLE_GattServer_AddService with mbed-cli and drag resulting .hex file to the NRF52_DK Jlink connection. Use NRF Connect to connect with Heartrate device. Will disconnect immediately.

paul-szczepanek-arm, 2021-01-04:
I notice it's also an iphone. Any chance I could have some more logs? Ideally something lower level, from ios or failing that if you could print the connection params from mbed side. From this amount of information it's hard to say what could cause a timeout. Do you have the mbed device connected via serial to make sure the application is running OK there?

paul-szczepanek-arm, 2021-01-05:
If it starts advertising it means it thinks the other side disconnected, can you print the disconnect reason on the mbed board?

apcurtiss, 2021-01-05:
Yes I've already done that, it's 0x08 (timeout), same as in the other issue.

pan-, 2021-01-05:
@apcurtiss What surprises me is it disconnects after sending an MTU request and the MTU size requested by the mobile is 527. The maximum is supposed to be 517...

Which model of iPhone do you use ?

apcurtiss, 2021-01-05:
@pan- Hm interesting... It's an iPhone 11 running iOS 14.3.

pan-, 2021-01-05:
It works fine on iPhone 8 😭 and Mac. Would it be possible for you to put traces in [attsProcMtuReq](https://github.com/ARMmbed/mbed-os/blob/4c4295225ec17f9485c61c72588a02f0d35d6b31/connectivity/FEATURE_BLE/libraries/cordio_stack/ble-host/sources/stack/att/atts_proc.c#L238) and tell us if the server is trying [to send an error](https://github.com/ARMmbed/mbed-os/blob/4c4295225ec17f9485c61c72588a02f0d35d6b31/connectivity/FEATURE_BLE/libraries/cordio_stack/ble-host/sources/stack/att/atts_proc.c#L254), What is the size of the [local MTU](https://github.com/ARMmbed/mbed-os/blob/4c4295225ec17f9485c61c72588a02f0d35d6b31/connectivity/FEATURE_BLE/libraries/cordio_stack/ble-host/sources/stack/att/atts_proc.c#L270) computed and if the [allocation of the response](https://github.com/ARMmbed/mbed-os/blob/4c4295225ec17f9485c61c72588a02f0d35d6b31/connectivity/FEATURE_BLE/libraries/cordio_stack/ble-host/sources/stack/att/atts_proc.c#L273) fail or not. Then post the result in this thread.

apcurtiss, 2021-01-05:
Sure thing. And for what it's worth in the meantime, here are the phone logs from the BLE_GattServer_CharacteristicUpdates program, which _does_ work with my iPhone. 

[GattServer_CharacteristicUpdates BLE Logs.txt](https://github.com/ARMmbed/mbed-os-example-ble/files/5771970/GattServer_CharacteristicUpdates.BLE.Logs.txt)

pan-, 2021-01-05:
That's a long shot but can you try to enable [LL_OP_MODE_FLAG_SLV_DELAY_LLCP_STARTUP](https://github.com/ARMmbed/mbed-os/blob/4c4295225ec17f9485c61c72588a02f0d35d6b31/connectivity/FEATURE_BLE/libraries/TARGET_CORDIO_LL/stack/controller/sources/ble/lmgr/lmgr_main.c#L74) ?

apcurtiss, 2021-01-05:
^^ just to clarify, when it was hit the attProcMtuReq function completed successfully and didn't try to send an error out.

pan-, 2021-01-06:
Thanks for trying @apcurtiss , these are difficult problems to solve without the hardware available to reproduce them. 
Both ends reporting a timeout as the disconnection reason is surprising to me, as well as the Characteristic update example working and not the hearthrate one. 

Are you experiencing incorrect behavior on other examples ?

jrobeson, 2021-05-24:
I can replicate this on an android based phone with another nrf52833 based board.

jrobeson, 2021-05-25:
I ended up finding this issue because I had the exact same problem with my own service, so i figured I'd start with the upstream examples to figure out if the problem was similar.

I migrated my own code to use mbed-ble-utils `ble_process.h`  (with the queue `dispatch_forever` call commented out and now it works again.

I wonder if it has to do with advertising handle code.

`gap.setAdvertisingPayload(ble::LEGACY_ADVERTISING_HANDLE, _adv_data_builder.getAdvertisingData());`
vs:
`gap.setAdvertisingPayload(_adv_handle, _adv_data_builder.getAdvertisingData());`