ARMmbed/mbed-os-example-ble #364: Mbed CLI 2 not linking BlueNrg HCI drivers

boraozgen, 2021-03-24:
### Description of defect

When building the examples with Mbed CLI 2 for DISCO_L562QE, I noticed that the BlueNrg HCI drivers do not get linked, resulting in a `Please provide an implementation for the HCI driver` error on initialization.

I then noticed that `mbed-ble-blue_nrg` CMake library does not get linked anywhere. I added this to the main CMake file:

`target_link_libraries(${APP_TARGET} PRIVATE mbed-ble-blue_nrg)`

and it started working. Of course this should be done in the Mbed CMake files according to the target configuration.

Otherwise CLI 2 builds for NRF52_DK without issues (had some problems with path lengths but it is Windows specific).

#### Target(s) affected by this defect ?

DISCO_L562QE

#### Toolchain(s) (name and version) displaying this defect ?

GCC for arm-none-eabi 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools 7.5.0

#### How is this defect reproduced ?

Try building + running an example for DISCO_L562QE with Mbed CLI 2.

jeromecoutant, 2021-03-29:
Quick check on my side:
Everythink seems OK...

````
xxx/mbed-os-example-ble/BLE_Advertising (master)
$ mbedtools compile -m DISCO_L562QE -t GCC_ARM
...
-- built: xxx/mbed-os-example-ble/BLE_Advertising/cmake_build/DISCO_L562QE/develop/GCC_ARM/BLE_Advertising.bin
-- built: xxx/mbed-os-example-ble/BLE_Advertising/cmake_build/DISCO_L562QE/develop/GCC_ARM/BLE_Advertising.hex
| Module               |           .text |       .data |          .bss |
|----------------------|-----------------|-------------|---------------|
| [fill]               |       252(+252) |       3(+3) |       47(+47) |
| [lib]\c.a            |     7312(+7312) | 2108(+2108) |       89(+89) |
| [lib]\gcc.a          |       956(+956) |       0(+0) |         0(+0) |
| [lib]\misc           |       188(+188) |       4(+4) |       28(+28) |
| [lib]\stdc++.a       |           0(+0) |       0(+0) |         0(+0) |
| mbed-os\cmsis        |     7952(+7952) |   168(+168) |   6212(+6212) |
| mbed-os\connectivity |   80478(+80478) |   241(+241) |   4762(+4762) |
| mbed-os\drivers      |       786(+786) |       0(+0) |         0(+0) |
| mbed-os\events       |     1258(+1258) |       0(+0) |         0(+0) |
| mbed-os\hal          |     1666(+1666) |       8(+8) |     114(+114) |
| mbed-os\platform     |     5708(+5708) |   260(+260) |     424(+424) |
| mbed-os\rtos         |       184(+184) |       0(+0) |         8(+8) |
| mbed-os\targets      |   12448(+12448) |       8(+8) |   1120(+1120) |
| source\main.cpp.obj  |     1296(+1296) |       0(+0) |       92(+92) |
| Subtotals            | 120484(+120484) | 2800(+2800) | 12896(+12896) |
Total Static RAM memory (data + bss): 15696(+15696) bytes
Total Flash memory (text + data): 123284(+123284) bytes
````

boraozgen, 2021-03-30:
The issue isn't that it doesn't build. It builds without issues, albeit without linking the HCI driver, which gives an error in runtime:

```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: _hci_driver != nullptr
Location: 0x8003A79
File: C:/mbed-os-example-ble/mbed-os/connectivity/FEATURE_BLE/source/cordio/source/BLEInstanceBaseImpl.cpp+466
Error Value: 0x0
Current Thread: main Id: 0x20001530 Entry: 0x8002425 StackSize: 0x1000 StackMem: 0x20002300 SP: 0x200031DC
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=TARGET_NAME
-- MbedOS Error Info --
Please provide an implementation for the HCI driver
```

You can also search for the `mbed-ble-blue_nrg` CMake target and see that it is not linked anywhere.

jeromecoutant, 2021-03-30:
Issue reproduced

closed 2021-04-14 by 0xc0170