ARMmbed/mbed-os-example-ble #365: Periodic advertisement example issue on WB55

pan-, 2021-03-29:
### Description of defect

From https://github.com/ARMmbed/mbed-os-example-ble/issues/355#issuecomment-805578819

They do not compile on GCC 9.3.1. I tried using the online compiler with the periodic advertisement example, while it compiles i can't get the Bluetooth to work properly (on a ST WB55).

#### Target(s) affected by this defect ?

ST WB55

#### Toolchain(s) (name and version) displaying this defect ?

GCC

#### How is this defect reproduced ?

Try to compile the application with GCC 9.6.3. It does not compile. 
Try to execute the application compiled with the online compiler: It doesn't work.

pan-, 2021-03-29:
@Arjun765 Is it a fair description of your issue ?

jeromecoutant, 2021-03-29:
@pan- 
Did you try on your side...?

jeromecoutant, 2021-03-29:
Then as explained in https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_PeriodicAdvertising
Devices must support extended advertising and periodic advertising (Bluetooth version 5+).

paul-szczepanek-arm, 2021-03-29:
Thank you @jeromecoutant.

@Arjun765 we now have a handy new example:
https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_SupportedFeatures
which lists the available features.

closed 2021-03-29 by paul-szczepanek-arm

pan-, 2021-03-30:
@jeromecoutant I opened the issue to keep track of it. It is expected extended advertising won't work on WB55. 
Compilation is fine though.

Arjun765, 2021-03-30:
`BLE_SupportedFeatures` works on a WB55. Tried the `BLE_Advertising` on a nrf52840, while i can see it advertising as "BATTERY" i can't connect to it. The nrf mobile app says "Error 133, GATT error"

paul-szczepanek-arm, 2021-03-30:
That is correct. It is advertising with non-connectable advertising. You can connect to BLE_GattServer_CharacteristicWrite for example.