ARMmbed/mbed-os-example-ble #366: Relying n event queue seems to be a really ad idea

idea--list, 2021-04-06:
### Description of defect

Based on the BLE_GattServer_AddService example i wrote a really basic signal generator that streams values to another BLE device after a connection is established. The intention of this was to see how many packets get lost and how reliable is BLE API in general. The nRF Connect (and similar) apps are simply not fit for such evaluations... so i designed an app for scoping the BLE stream with App Inventor.

Major conclusions:

- Mbed BLE API can handle up to ca. 30ms connection intervals (~ 30Hz update rate).
- If the connection interval is shorter, the BLE signal will begin to fall apart (above 40Hz it becomes extremely apparent).
- If you disconnect and reconnect the devices then the data transmission rate doubles after each reconnection (i guess this is because of the nature of `event_queue.call_every()`. Until now i could not find any means to circumvent that...except for `system_reset()` which is not a viable option in real world applications)

The last 2 points simply render Mbed BLE API completely unusable for almost anything beyond unit tests on some developers workbench that should only run for some seconds. I mean in any real application chances are high that the BLE connection will get lost and the devices will reconnect...which will cause the BLE signal ripple apart.

I wonder what i might be missing, so any hint is welcome.

#### Target(s) affected by this defect ?
Artemis Thing Plus

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.15

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS V6.9

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V1.4

#### How is this defect reproduced ?
Code for the BLE signal generator:
```
/* mbed Microcontroller Library
 * Copyright (c) 2006-2015 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include "ble/BLE.h"
#include "ble/gap/Gap.h"
#include "ble/services/HeartRateService.h"
#include "pretty_printer.h"
#include <cstdint>
#include <events/mbed_events.h>

#define UPDATEINTV 100ms  //defines data update interval

using namespace std::literals::chrono_literals;

uint8_t signalType = 0; // 0 for saw   1 for Zig-Zag   2 for sine wave

// Sine wave values
uint8_t sine256[] = {
    127, 130, 133, 136, 139, 143, 146, 149, 152, 155, 158, 161, 164, 167, 170,
    173, 176, 178, 181, 184, 187, 190, 192, 195, 198, 200, 203, 205, 208, 210,
    212, 215, 217, 219, 221, 223, 225, 227, 229, 231, 233, 234, 236, 238, 239,
    240, 242, 243, 244, 245, 247, 248, 249, 249, 250, 251, 252, 252, 253, 253,
    253, 254, 254, 254, 254, 254, 254, 254, 253, 253, 253, 252, 252, 251, 250,
    249, 249, 248, 247, 245, 244, 243, 242, 240, 239, 238, 236, 234, 233, 231,
    229, 227, 225, 223, 221, 219, 217, 215, 212, 210, 208, 205, 203, 200, 198,
    195, 192, 190, 187, 184, 181, 178, 176, 173, 170, 167, 164, 161, 158, 155,
    152, 149, 146, 143, 139, 136, 133, 130, 127, 124, 121, 118, 115, 111, 108,
[... 164 lines cut ...]
  UUID _heartrate_uuid;
  uint8_t _signal_value;
  HeartRateService _heartrate_service;

  uint8_t _adv_buffer[ble::LEGACY_ADVERTISING_MAX_SIZE];
  ble::AdvertisingDataBuilder _adv_data_builder;
};

/* Schedule processing of events from the BLE middleware in the event queue. */
void schedule_ble_events(BLE::OnEventsToProcessCallbackContext *context) {
  event_queue.call(Callback<void()>(&context->ble, &BLE::processEvents));
}

int main() {
  BLE &ble = BLE::Instance();
  ble.onEventsToProcess(schedule_ble_events);

  HeartrateDemo demo(ble, event_queue);
  demo.start();

  return 0;
}
```

Import [this project](https://drive.google.com/file/d/1ddV43s_znogzVgeJOfpMYxKvvr6vR7w8/view?usp=sharing) into [App Inventor](http://ai2.appinventor.mit.edu/) and use your phone as a scope.

pan-, 2021-04-06:
Thanks for sharing @idea--list . I don't think the issue has anything to do with the event queue which should be fast enough to handle any kind of transfer at the highest rate offered by BLE (every 6 ms). 
In the example posted above, the issue seems to lie in data queuing which is unconditional of the link configuration. By default Android connection interval is set to 50ms. If you push new data every 30ms of course it will saturate the transfer queue. 

My recommendation is to split sampling rate from transfer rate: At a regular interval set locally the data you want to send. Then use [`GattServer::EventHandle::onDataSent`](https://github.com/ARMmbed/mbed-os/blob/73896715e6202e187ba624fc3853299e73390f19/connectivity/FEATURE_BLE/include/ble/GattServer.h#L131) to send new data through the BLE link: when it is called, it is safe to send new data. You can use a [CircularBuffer](https://github.com/ARMmbed/mbed-os/blob/master/platform/include/platform/CircularBuffer.h) If you want to store more than one sample.

paul-szczepanek-arm, 2021-04-06:
Store samples and send them in a batch. The frequency of updates doubles after reconnection because you call ```_event_queue.call_every(UPDATEINTV, [this] { update_sensor_value(); });``` a second time (the one from before is still running).

We're happy to help but this is not the place to ask these questions. This repository is for bugs with example code. If you have a general mbed-os bug to report please use the mbed-os repo. If you have programming questions then the forum is the better venue and will gather a wider audience.

closed 2021-04-06 by paul-szczepanek-arm