ARMmbed/mbed-os-example-ble #379: Some examples don't expose their list of services

jrobeson, 2021-06-02:
### Description of defect

BLE_GattServer_CharacteristicUpdates, and BLE_GattServer_CharacteristicWrite (at least)
don't expose the list of services they offer, such as via `Complete list of 16-it Service UUIDs`.

However that GattServer_addService example still does. The difference seems to be that this one is still
or or less the old way (pre `mbed-os-ble-utils`).

Can someone else reproduce this behavior?
#### Target(s) and toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of the example and mbed-os are you using (tag or sha) ?
6.11

#### How is this defect reproduced ?
1. Flash an example
2. Open up NRFConnect (or equivalent)
3. Choose the device
3. View the details of the advertisement (in NRFConnect click "More")
4. 
#### Please attach a file containing the log with traces enabled.
I'm afraid i can't access to a board that has serial output atm, so no traces here.

paul-szczepanek-arm, 2021-06-02:
They don't. There's no such requirement for advertising payload to contain service uuids.

jrobeson, 2021-06-02:
There isn't a requirement, but it does mean that apis that allow you to filter on scan won't work.

paul-szczepanek-arm, 2021-06-02:
I can check but I think I removed the uuids because the examples does not actually run the service. If there is a genuine service there I can put the uuid back.

paul-szczepanek-arm, 2021-06-17:
OK, this was done to simplify the flow of the example. Examples are purely for educational reasons so to avoid cluttering the code with things that the example doesn't intend to teach those elements are omitted. Hence some examples use a ready-made process and only feature code relevant to the part of BLE API they intend to teach. For filtering you can can filter advertising by name. All examples have a name present in the advertising payload.

paul-szczepanek-arm, 2021-06-18:
Closing as I will not be adding the services list back in.

closed 2021-06-18 by paul-szczepanek-arm