ARMmbed/mbed-os-example-ble #381: BLE_Advertising example stack overflow nRF52 if debugging with MBED_TICKLESS is set

pearson, 2021-06-15:
### Description of defect

The Mbed BLE_Advertising example fails with a stack overflow on an nRF52832 in debug profile if the default `MBED_TICKLESS` parameter is set in the `MCU_NRF52832` section `mbed-os\targets\targets.json`.

```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x4E25
Error Value: 0x1
Current Thread: <unnamed> Id: 0x20000D08 Entry: 0x1CF1C StackSize: 0x20000 StackMem: 0x20002408 SP: 0x2000FECC
For more info, visit: https://mbed.com/s/error?error=0x80020125&tgt=TARGET_NAME
-- MbedOS Error Info --
```

This problem may be related to the `MBED_TICKLESS` debugging behavior described in [issue #14766 in mbed-os](https://github.com/ARMmbed/mbed-os/issues/14766).

#### Target(s) affected by this defect ?
nRF52832, possibly other nRF devices

#### Toolchain(s) (name and version) displaying this defect ?
The defect appears with either the command line tools or if built through PlatformIO:
- Mbed CLI 2 v7.20.0, arm-none-eabi 10.2.1
- PlatformIO, v2.3.2, arm-none-eabi-gcc 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?
- For the command line tools, the version pulled with `mbed-tools.exe import mbed-os-example-ble`
   - Commit SHA: 39644781a280bd732e58cb75686a40b5f57b07b3
- For PlatformIO, it's unclear which version is in use.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
- Mbed CLI 2 v7.20.0
- For PlatformIO, it's unclear which version is in use.

#### How is this defect reproduced ? 
- Import the BLE_Advertising example with `mbed-tools.exe import mbed-os-example-ble ble`
- Configure BLE_Advertising with `mbed-tools.exe configure -m NRF52_DK -t GCC_ARM`
- Compile BLE_Advertising with `mbed-tools.exe compile -m nRF52_DK -t GCC_ARM -b debug --clean`
- Open a serial terminal
- Start a debug session with any debugger. (Tested with VS Code, with and without PlatformIO, and SEGGER Ozone.)
- Allow execution to continue
- The MbedOS Error shown above will appear in the serial terminal

The problem will NOT appear if:
- `MBED_TICKLESS` is removed from the `MCU_NRF52832` section of `mbed-os\targets\targets.json` and the project is reconfigured
- The project is build in release mode with `mbed-tools.exe compile -m nRF52_DK -t GCC_ARM -b release --clean` with the `MBED_TICKLESS` parameter intact as is the default.

pearson, 2021-06-15:
@ciarmcom , you're welcome! 

I seem to be having a lot of issues with the nRF52832 and Mbed. I'll be posting another shortly regarding the BLE_PeriodicAdvertising example.

paul-szczepanek-arm, 2021-06-17:
I don't think you're problems are anything to do with examples. If you're having issues with the stack being too small, try doubling the stack size.

pearson, 2021-06-17:
Hi @paul-szczepanek-arm . I'm going to close this issue for now as I can no longer replicate the problem.

It's really odd... I tested my steps to reproduce (and edited them yesterday), and I could reproduce it. 

Sorry about that. I'll reopen the problem if I can reproduce it again. That said, [issue #14766 in mbed-os](https://github.com/ARMmbed/mbed-os/issues/14766) with the nRF52832, `MBED_TICKLESS`, `ThisThread::sleep_for()`, and debugging is still reproducible on more than one nRF52832.

closed 2021-06-17 by pearson