ARMmbed/mbed-os-example-ble #392: Definition of 'GattServer' device to search for

kr0sh1, 2021-11-27:
In the docu of the 'BLE_GattClient_CharacteristicUpdates' example it mentions that it's hardcoded (presumably) to search for a device called 'GattServer'.

Where is this defined and how can it be changed?

noonfom, 2021-11-29:
Hi @kr0sh1, that appears to be hardcoded inside the [`get_device_name()`](https://github.com/ARMmbed/mbed-os-ble-utils/blob/0da6cf31bd6688e4a3ecdf98b356e4767f3248c4/gatt_server_process.h#L33) function of the `GattServerProcess` class.

Editing it seems nontrivial. @paul-szczepanek-arm could this be a private member with a public setter?

kr0sh1, 2021-11-29:
@noonfom perfect, thanks for that!

paul-szczepanek-arm, 2021-11-29:
If you just want to change it in the example you can just edit the string here https://github.com/ARMmbed/mbed-os-ble-utils/blob/0da6cf31bd6688e4a3ecdf98b356e4767f3248c4/gatt_client_process.h#L46 to change what device name it's looking for.